OSCPU/yosys-sta #2: Failed to generate evaluation result of lab6 in dlco course

junyu33, 2023-08-18:
Here is the lab link: [lab6](https://nju-projectn.github.io/dlco-lecture-note/exp/06.html)

The RTL code is:

```verilog
module bin16seg(
  input [3:0] bin,
  output [6:0] seg
);
  
  assign seg = (bin == 4'b0000) ? 7'b0000001 :
               (bin == 4'b0001) ? 7'b1001111 :
               (bin == 4'b0010) ? 7'b0010010 :
               (bin == 4'b0011) ? 7'b0000110 :
               (bin == 4'b0100) ? 7'b1001100 :
               (bin == 4'b0101) ? 7'b0100100 :
               (bin == 4'b0110) ? 7'b0100000 :
               (bin == 4'b0111) ? 7'b0001111 :
               (bin == 4'b1000) ? 7'b0000000 :
               (bin == 4'b1001) ? 7'b0000100 :
               (bin == 4'b1010) ? 7'b0001000 :
               (bin == 4'b1011) ? 7'b1100000 :
               (bin == 4'b1100) ? 7'b0110001 :
               (bin == 4'b1101) ? 7'b1000010 :
               (bin == 4'b1110) ? 7'b0110000 :
               (bin == 4'b1111) ? 7'b0111000 :
               7'b1111111;

endmodule

module lfsr8bit(
  input [7:0] init,
  input clk,
  input rst,
  output reg [7:0] q
);
  
  always @(posedge clk or posedge rst) begin
    if (rst) begin
      q <= init;
    end else begin
      q <= {q[4] ^ q[3] ^ q[2] ^ q[0], q[7:1]};
    end
  end

endmodule

module lab6( // use button of clk, and map the output of lfsr8bit to bin16seg
  input [7:0] init,
  input clk,
  input rst,
  output [6:0] hex0,
  output [6:0] hex1
);
  
  wire [7:0] q;
  lfsr8bit lfsr(init, clk, rst, q);

  bin16seg seg1(q[3:0], hex0);
  bin16seg seg2(q[7:4], hex1);

endmodule
```

and the sdc file is:

```sdc
set clk_port_name clk
set CLK_FREQ_MHZ 500
if {[info exists env(CLK_FREQ_MHZ)]} {
  set CLK_FREQ_MHZ $::env(CLK_FREQ_MHZ)
} else {
  puts "Warning: Environment CLK_FREQ_MHZ is not defined. Use $CLK_FREQ_MHZ MHz by default."
}
set clk_io_pct 0.2

set clk_port [get_ports $clk_port_name]
create_clock -name core_clock -period [expr 1000 / $CLK_FREQ_MHZ] $clk_port
```

Here is the command and the result:

```zsh
~/Desktop/ysyx/ysyx-workbench/yosys-sta master                                       
> make sta DESIGN=lab6 SDC_FILE=~/Desktop/ysyx/ysyx-workbench/yosys-sta/example/gcd.sdc RTL_FILES=~/Desktop/ysyx/ysyx-workbench/npc/vsrc/lab6.v CLK_FREQ_MHZ=100
cd bin && LD_LIBRARY_PATH=lib/ ./iSTA /home/junyu33/Desktop/ysyx/ysyx-workbench/yosys-sta/sta.tcl lab6 ~/Desktop/ysyx/ysyx-workbench/yosys-sta/example/gcd.sdc /home/junyu33/Desktop/ysyx/ysyx-workbench/yosys-sta/result/lab6-100MHz/lab6.netlist.v
***************************
  _  _____  _____   ___
 (_)/  ___||_   _| / _ \
  _ \ `--.   | |  / /_\ \
 | | `--. \  | |  |  _  |
 | |/\__/ /  | |  | | | |
 |_|\____/   \_/  \_| |_/
***************************
WELCOME TO iSTA TCL-shell interface.

I0818 10:23:56.980780  7597 Sta.cc:300] read verilog file /home/junyu33/Desktop/ysyx/ysyx-workbench/yosys-sta/result/lab6-100MHz/lab6.netlist.v start
I0818 10:23:56.984899  7597 Sta.cc:304] read verilog end
I0818 10:23:56.984970  7597 Sta.cc:270] load lib start
I0818 10:23:56.987181  7629 Liberty.cc:2720] load liberty file /home/junyu33/Desktop/ysyx/ysyx-workbench/yosys-sta/nangate45/lib/merged.lib
I0818 10:23:57.098194  7629 LibertyParse.y:66] load liberty file /home/junyu33/Desktop/ysyx/ysyx-workbench/yosys-sta/nangate45/lib/merged.lib success.
I0818 10:23:57.575963  7597 Sta.cc:289] load lib end
I0818 10:23:57.576033  7597 Sta.cc:331] link design start
I0818 10:23:57.576040  7597 VerilogReader.cc:847] flatten module lab6 start
I0818 10:23:57.576066  7597 VerilogReader.cc:863] flatten module lfsr8bit inst lfsr
I0818 10:23:57.576213  7597 VerilogReader.cc:863] flatten module bin16seg inst seg1
I0818 10:23:57.576370  7597 VerilogReader.cc:863] flatten module bin16seg inst seg2
I0818 10:23:57.576522  7597 VerilogReader.cc:874] flatten module lab6 end
I0818 10:23:57.576712  7597 Sta.cc:438] liberty cell $_ALDFF_PP_ is not exist.
I0818 10:23:57.576720  7597 Sta.cc:438] liberty cell $_ALDFF_PP_ is not exist.
I0818 10:23:57.576725  7597 Sta.cc:438] liberty cell $_ALDFF_PP_ is not exist.
I0818 10:23:57.576750  7597 Sta.cc:438] liberty cell $_ALDFF_PP_ is not exist.
I0818 10:23:57.576756  7597 Sta.cc:438] liberty cell $_ALDFF_PP_ is not exist.
I0818 10:23:57.576761  7597 Sta.cc:438] liberty cell $_ALDFF_PP_ is not exist.
I0818 10:23:57.576767  7597 Sta.cc:438] liberty cell $_ALDFF_PP_ is not exist.
I0818 10:23:57.576772  7597 Sta.cc:438] liberty cell $_ALDFF_PP_ is not exist.
I0818 10:23:57.578671  7597 Sta.cc:624] link design end
I0818 10:23:57.578686  7597 Sta.cc:172] read sdc /home/junyu33/Desktop/ysyx/ysyx-workbench/yosys-sta/example/gcd.sdc start
I0818 10:23:57.579398  7597 CmdCreateClock.cc:115] create clock core_clock for pin/port: clk
I0818 10:23:57.579412  7597 Sta.cc:185] read sdc end
I0818 10:23:57.579443  7597 StaBuildGraph.cc:310] build graph start
I0818 10:23:57.580286  7597 StaBuildGraph.cc:212] net clk has no driver.
I0818 10:23:57.580324  7597 StaBuildGraph.cc:212] net init[0] has no driver.
I0818 10:23:57.580330  7597 StaBuildGraph.cc:212] net init[1] has no driver.
I0818 10:23:57.580349  7597 StaBuildGraph.cc:212] net init[2] has no driver.
I0818 10:23:57.580374  7597 StaBuildGraph.cc:212] net init[3] has no driver.
I0818 10:23:57.580379  7597 StaBuildGraph.cc:212] net init[4] has no driver.
I0818 10:23:57.580384  7597 StaBuildGraph.cc:212] net init[5] has no driver.
I0818 10:23:57.580389  7597 StaBuildGraph.cc:212] net init[6] has no driver.
I0818 10:23:57.580410  7597 StaBuildGraph.cc:212] net init[7] has no driver.
I0818 10:23:57.580415  7597 StaBuildGraph.cc:212] net q[0] has no driver.
I0818 10:23:57.580421  7597 StaBuildGraph.cc:212] net q[1] has no driver.
I0818 10:23:57.580426  7597 StaBuildGraph.cc:212] net q[2] has no driver.
I0818 10:23:57.580431  7597 StaBuildGraph.cc:212] net q[3] has no driver.
I0818 10:23:57.580436  7597 StaBuildGraph.cc:212] net q[4] has no driver.
I0818 10:23:57.580441  7597 StaBuildGraph.cc:212] net q[5] has no driver.
I0818 10:23:57.580446  7597 StaBuildGraph.cc:212] net q[6] has no driver.
I0818 10:23:57.580451  7597 StaBuildGraph.cc:212] net q[7] has no driver.
I0818 10:23:57.580456  7597 StaBuildGraph.cc:212] net rst has no driver.
I0818 10:23:57.580600  7597 StaBuildGraph.cc:327] build graph end
I0818 10:23:57.580615  7597 Sta.cc:1959] update timing start
I0818 10:23:57.580734  7597 StaApplySdc.cc:648] apply sdc start
I0818 10:23:57.580765  7597 StaApplySdc.cc:675] apply sdc end
I0818 10:23:57.580772  7597 StaClockPropagation.cc:264] ideal clock propagation start
I0818 10:23:57.580781  7597 StaClockPropagation.cc:334] ideal clock propagation end
I0818 10:23:57.580787  7597 StaCheck.cc:129] found loop fwd start
I0818 10:23:57.580807  7597 StaCheck.cc:148] found loop fwd end
I0818 10:23:57.580813  7597 StaCheck.cc:150] found loop bwd start
I0818 10:23:57.580829  7597 StaCheck.cc:172] found loop bwd end
I0818 10:23:57.580835  7597 StaSlewPropagation.cc:265] slew propagation start
F0818 10:23:57.582027  7661 StaSlewPropagation.cc:208] slew propgation start point seg1/_74_:A is not input port or output pin.
*** Check failure stack trace: ***
F0818 10:23:57.582072  7633 StaSlewPropagation.cc:208] slew propgation start point seg1/_73_:A is not input port or output pin.F0818 10:23:57.582072  7632 StaSlewPropagation.cc:208] slew propgation start point seg1/_72_:A is not input port or output pin.F0818 10:23:57.582077  7636 StaSlewPropagation.cc:208] slew propgation start point seg2/_74_:A is not input port or output pin.F0818 10:23:57.582118  7640 StaSlewPropagation.cc:208] slew propgation start point seg2/_73_:A is not input port or output pin.F0818 10:23:57.582118  7639 StaSlewPropagation.cc:208] slew propgation start point seg2/_72_:A is not input port or output pin.
*** Check failure stack trace: ***
make: *** [Makefile:26: /home/junyu33/Desktop/ysyx/ysyx-workbench/yosys-sta/result/lab6-100MHz/lab6.rpt] Aborted (core dumped)
```

I've checked the README and ensure these:

> 在`RTL_FILES`的文件中必须包含一个名为`DESIGN`的module

The compile command and top module are both `lab6`.
> sdc文件中的时钟端口名称需要与设计文件保持一致, 具体内容可参考样例设计GCD中的相应文件

The sdc file and my RTL both use `clk` for input.

simintao, 2023-08-19:
[image]
同学请加微信群交流

simintao, 2023-08-19:
还需要提供netlist文件给我

simintao, 2023-08-19:
新版本9eff94c1885f9260ffad950ae2c8e5ff96768672已经接近crash

sashimi-yzh, 2023-08-20:
已修复, 并修改为通过`wget`拉取最新版本的iSTA, 请移除并重新克隆本项目, 尝试能否通过测试.

junyu33, 2023-08-20:
OK, problem solved.

closed 2023-08-20 by junyu33