OSCPU/yosys-sta #5: module含有未使用的端口会导致yosys-sta报出get port

out-of-order55, 2024-04-28:
当我使用commit号为8deb008e3f6413a7a07426ff62b46d2ce840ce96的yosys-sta时,出现当模块或子模块内部有未使用的端口时,yosys-sta爆出以下错误
[image]
当注释掉未使用端口后,yosys-sta可正常运行
我的测试代码如下
`module test (
    input               clk,
    input               a,
    input               b,
    output              c
);
    reg temp;
    always @(posedge clk) begin
        temp <= a;
    end
    assign c = temp;
endmodule`
b为未使用的端口,sdc文件使用的example的gcd.sdc

out-of-order55, 2024-04-29:
[result.zip](https://github.com/OSCPU/yosys-sta/files/15144465/result.zip)
这是综合后的result

sashimi-yzh, 2024-07-12:
我们已经更新了iEDA的版本. 请尝试已下命令:
```bash
cd yosys-sta
git pull origin master
rm -rf bin
make init
# try to generate the timing report of your design
```

closed 2024-08-25 by sashimi-yzh