OSCPU/yosys-sta #6: 顶层未使用信号在fix-fanout时出现error导致生成空的时序分析报告

xixi-shredp, 2024-05-01:
`make sta` 后，yosys可以正常运行，输出result当中也有网表相关的内容，但关于时序分析报告全是空的。  

```
Generate the report at 2024-04-30T23:59:26, GitVersion: d9c76846ce6e28304fe193503daa05d7efa8c4cc.
+----------+-------------+------------+------------+---------------+------+-------+-----------+
| Endpoint | Clock Group | Delay Type | Path Delay | Path Required | CPPR | Slack | Freq(MHz) |
+----------+-------------+------------+------------+---------------+------+-------+-----------+
+-------+------------+-----+
| Clock | Delay Type | TNS |
+-------+------------+-----+

```

## Error 

根据`yosys.log`来看，yosys本身综合似乎并没有出现什么异常，相反`sta.log`当中出现了大量的`has no clock data`。  

```
E20240430 23:59:26.152082 36756 StaAnalyze.cc:302] end vertex clint/_1249_:D has no clock data.
E20240430 23:59:26.152084 36756 StaAnalyze.cc:302] end vertex clint/_1249_:D has no clock data.
E20240430 23:59:26.152088 36756 StaAnalyze.cc:302] end vertex clint/_1250_:D has no clock data.
E20240430 23:59:26.152091 36756 StaAnalyze.cc:302] end vertex clint/_1250_:D has no clock data.
E20240430 23:59:26.152094 36756 StaAnalyze.cc:302] end vertex clint/_1251_:D has no clock data.
E20240430 23:59:26.152098 36756 StaAnalyze.cc:302] end vertex clint/_1251_:D has no clock data.
E20240430 23:59:26.152101 36756 StaAnalyze.cc:302] end vertex clint/_1252_:D has no clock data.
E20240430 23:59:26.152107 36756 StaAnalyze.cc:302] end vertex clint/_1252_:D has no clock data.
E20240430 23:59:26.152112 36756 StaAnalyze.cc:302] end vertex clint/_1211_:D has no clock data.
E20240430 23:59:26.152117 36756 StaAnalyze.cc:302] end vertex clint/_1211_:D has no clock data.
E20240430 23:59:26.152123 36756 StaAnalyze.cc:302] end vertex clint/_1212_:D has no clock data.
E20240430 23:59:26.152128 36756 StaAnalyze.cc:302] end vertex clint/_1212_:D has no clock data.
E20240430 23:59:26.152133 36756 StaAnalyze.cc:302] end vertex clint/_1213_:D has no clock data.
E20240430 23:59:26.152138 36756 StaAnalyze.cc:302] end vertex clint/_1213_:D has no clock data.
E20240430 23:59:26.152143 36756 StaAnalyze.cc:302] end vertex clint/_1214_:D has no clock data.
...
```

从生成的.v文件一级一级回溯后发现问题发生在`$(DESIGN).netlist.fixed.v`：  
```
//Generate the verilog at 2024-04-30T23:59:23
module DESIGN (
clock
);

input clock ;

LOGIC1_X1 _02_ ( .Z( ) );
LOGIC0_X1 _03_ ( .Z( ) );
BUF_X1 _04_ ( .A( ), .Z( ) );
...
```

我发现经过`yosys/Makefile`下的`fix-fanout`这条rule之后得到的`$(NETLIST_FIXED_V)`文件当中的输入只有`clock`，而且在整个文件中没有再被引用过，所以查询过`$(RESULT_DIR)/fix-fanout.log`后发现出现了如下的错误信息：  

```
Read DEF file : /path/to/result/DESIGN-500MHz/DESIGN.syn.def
-----------------------
ERROR (DEFPARS-5500): Def parser has encountered an error in file /path/to/result/DESIGN-500MHz/DESIGN.syn.def at line 46141, on token +.
Problem can be syntax error on the def file or an invalid parameter name.
Double check the syntax on the def file with the LEFDEF Reference Manual.
**************************************************************
*************************Layout*******************************
```

于是在`DESIGN.syn.def`这个文件中找到了相应的代码：  

```
PINS 519 ;
 - clock + NET clock + DIRECTION INPUT  + USE SIGNAL

;
 - io_interrupt + NET  + DIRECTION INPUT  + USE SIGNAL

;
 - io_master_araddr[0] + NET io_master_araddr[0] + DIRECTION OUTPUT  + USE SIGNAL
 ...
```

`io_interrupt`是顶层的输入信号，但我并没有使用过该信号，其NET直接留空，对应源文件如下：  

```
module DESIGN(	// @[<stdin>:2225:10]
  input         clock,	// @[<stdin>:2226:11]
                reset,	// @[<stdin>:2227:11]
                io_interrupt,	// @[src/main/system.scala:27:14]
                io_master_awready,	// @[src/main/system.scala:27:14]
				...
```

## Try 
尝试去掉该信号后重新综合，可以得到完整的时序报告：  

```
Generate the report at 2024-05-01T10:40:24, GitVersion: d9c76846ce6e28304fe193503daa05d7efa8c4cc.
+------------------------------+-------------+------------+------------+---------------+-------+--------+-----------+
| Endpoint                     | Clock Group | Delay Type | Path Delay | Path Required | CPPR  | Slack  | Freq(MHz) |
+------------------------------+-------------+------------+------------+---------------+-------+--------+-----------+
| cpuCore/cpuWBU/pcReg/_463_:D | core_clock  | max        | 3.849f     | 1.961         | 0.000 | -1.888 | 257.211   |
| cpuCore/cpuWBU/pcReg/_465_:D | core_clock  | max        | 3.839f     | 1.961         | 0.000 | -1.878 | 257.832   |
| cpuCore/cpuWBU/pcReg/_490_:D | core_clock  | max        | 3.837f     | 1.961         | 0.000 | -1.876 | 257.968   |
| cpuCore/cpuEXU/_2213_:D      | core_clock  | min        | 0.125f     | 0.002         | 0.000 | 0.123  | NA        |
| cpuCore/cpuEXU/_2216_:D      | core_clock  | min        | 0.125f     | 0.002         | 0.000 | 0.123  | NA        |
| cpuCore/cpuEXU/_2214_:D      | core_clock  | min        | 0.125f     | 0.002         | 0.000 | 0.123  | NA        |
+------------------------------+-------------+------------+------------+---------------+-------+--------+-----------+

```

所以，请问yosys-sta对于顶层未使用的input信号是有什么特殊要求吗？另外，请问能否在中间`fix-fanout`出现error时中止综合过程，而不是继续运行产生一些预期之外的结果？

## Info
- iEDA版本号: Git version: bb7f026a1df9bffcf508087af682fe32931b950f
- Platform : ubuntu 22.04
## Src
rtl源文件，以及原始的分析报告文件和sdc文件如下：  
[src.zip](https://github.com/OSCPU/yosys-sta/files/15173171/src.zip)

0xharry, 2024-05-01:
同学你好，目前看来这个问题是iEDA对def文件生成的某些逻辑问题导致的，临时解决方案可以参考：
https://github.com/OSCC-Project/iEDA/blob/e400704b950c84db42a8e149821753baba5c5689/scripts/design/sky130_gcd/run_iEDA.sh#L26
使用sed对def文本进行修改

sashimi-yzh, 2024-05-01:
我这边使用yosys v0.23, 暂时没有遇到这个问题, 供参考.

xixi-shredp, 2024-05-02:
我将yosys更新到 v0.40 以后也确实没有这个问题了，之前 apt 只能下载到 v0.9 ，这个版本确实会存在上述问题，非常感谢!

closed 2024-05-02 by xixi-shredp

sashimi-yzh, 2024-05-02:
不过这个问题其实还是ieda的问题, 只不过新yosys综合出的结果没有触发到这个问题而已