OSCPU/yosys-sta #7: About the version of YOSYS

yzcccccccccc, 2024-05-08:
The `Readme` file instructs to install yosys with `apt install yosys`, which will only get a low version of yosys tools. This version of yosys still have some bugs and cannot handle specific HDL files.   

For example, I generated `ALU.sv` (see the attachment) from CHISEL, which means that there is little possibility that my source code is wrong. However, yosys tool will triggered a error as `unexpected token blablabla...`.  

One possible way to fix this is following the guide at YOSYS(<https://github.com/YosysHQ/yosys>) to install the latest version. Actually this works for me. :)

[ALU.zip](https://github.com/OSCPU/yosys-sta/files/15251212/ALU.zip)

sashimi-yzh, 2024-05-08:
Which version of chisel do you use? 3.x or above?

sashimi-yzh, 2024-05-09:
We have reproduced the issue.

You are right. This issue can only be fixed with a higher version of `yosys`. Besides installing the latest version of `yosys`, using ubuntu 24.04 is another solution.

closed 2025-04-28 by ChrisYzcc