OSCPU/yosys-sta #11: 我已经安装了libyaml-cpp.so.0.7，但是iEDA仍然无法找到该库

zyk007, 2024-08-11:
我已经安装了libyaml-cpp.so.0.7，但是iEDA仍然无法找到该库
`./bin/iEDA: error while loading shared libraries: libyaml-cpp.so.0.7: cannot open shared object file: No such file or directory`
具体情况如下：
```
len@len-Lenovo-XiaoXinAir-14ITL-2021:~/ysyx/ysyx-workbench/yosys-sta$ make sta
./bin/iEDA -script /home/len/ysyx/ysyx-workbench/yosys-sta/scripts/fix-fanout.tcl /home/len/ysyx/ysyx-workbench/yosys-sta/example/gcd.sdc /home/len/ysyx/ysyx-workbench/yosys-sta/result/gcd-500MHz/gcd.netlist.syn.v gcd /home/len/ysyx/ysyx-workbench/yosys-sta/result/gcd-500MHz/gcd.netlist.fixed.v 2>&1 | tee /home/len/ysyx/ysyx-workbench/yosys-sta/result/gcd-500MHz/fix-fanout.log
./bin/iEDA: error while loading shared libraries: libyaml-cpp.so.0.7: cannot open shared object file: No such file or directory
./bin/iEDA -script /home/len/ysyx/ysyx-workbench/yosys-sta/scripts/sta.tcl /home/len/ysyx/ysyx-workbench/yosys-sta/example/gcd.sdc /home/len/ysyx/ysyx-workbench/yosys-sta/result/gcd-500MHz/gcd.netlist.fixed.v gcd 2>&1 | tee /home/len/ysyx/ysyx-workbench/yosys-sta/result/gcd-500MHz/sta.log
./bin/iEDA: error while loading shared libraries: libyaml-cpp.so.0.7: cannot open shared object file: No such file or directory
len@len-Lenovo-XiaoXinAir-14ITL-2021:~/ysyx/ysyx-workbench/yosys-sta$ whereis libyaml-cpp.so.0.7
libyaml-cpp.so.0: /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.6 /usr/local/lib/libyaml-cpp.so.0.7
len@len-Lenovo-XiaoXinAir-14ITL-2021:~/ysyx/ysyx-workbench/yosys-sta$ ldd ./bin/iEDA 
./bin/iEDA: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./bin/iEDA)
./bin/iEDA: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./bin/iEDA)
./bin/iEDA: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./bin/iEDA)
./bin/iEDA: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by ./bin/iEDA)
./bin/iEDA: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by ./bin/iEDA)
        linux-vdso.so.1 (0x00007ffcd15f1000)
        libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x00007f552c128000)
        libyaml-cpp.so.0.7 => not found
        libgomp.so.1 => /lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f552c0e6000)
        libtcl8.6.so => /lib/x86_64-linux-gnu/libtcl8.6.so (0x00007f552bf2a000)
        libunwind.so.8 => /lib/x86_64-linux-gnu/libunwind.so.8 (0x00007f552bf0d000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f552beef000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f552bd0d000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f552bbbe000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f552bba3000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f552b9b1000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f552e342000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f552b9ab000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f552b986000)
        liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f552b95d000)

```
环境：Ubuntu20.04
由于该环境下的apt工具链里面只有libyaml-cpp0.6，所以我这个库是直接从github下载源码通过cmake编译得到的，下载链接如下：https://github.com/jbeder/yaml-cpp/releases

希望有大佬帮忙看看，谢谢！

sashimi-yzh, 2024-08-11:
报错信息显示glibc的版本不一致，ubuntu的版本太低了，可以试试从源码编译iEDA https://gitee.com/oscc-project/iEDA.git

zyk007, 2024-08-13:
> 报错信息显示glibc的版本不一致，ubuntu的版本太低了，可以试试从源码编译iEDA https://gitee.com/oscc-project/iEDA.git

已解决，感谢解答！不过解决的过程有补充：
我尝试从源码编译iEDA，并且成功编译，只解决了libyaml-cpp0.6缺失的问题，仍然报缺少glibc的错误。
于是我参考下面的文章将gcc和libc0.6升级之后再去运行iEDA，最后才成功。
https://blog.csdn.net/huazhang_001/article/details/128828999
https://askubuntu.com/questions/575505/glibcxx-3-4-20-not-found-how-to-fix-this-error/786944

sashimi-yzh, 2024-08-13:
你是在ubuntu20.04下编译iEDA的吗? 如果是的话, 应该不会出现glibc版本不一致的错误

closed 2024-08-25 by sashimi-yzh