OSCPU/yosys-sta #12: 请问如何配置能够在综合时不实例化某些寄存器（例如性能计数器）？

WanderingNako, 2024-08-22:
我是一生一芯的学员，在做性能优化时，我想在综合时不实例化性能计数器，从而使得添加性能计数器不影响电路的面积和频率。请问应该如何配置？
我的RTL代码（DpicHalt.v）如下，其中inst_cnt, cycle_cnt, data_cnt, exec_cnt等是性能计数器，它们由各个模块负责采集，最后汇总在DpicHalt.v文件。
```
`define SYNTHESIS

module DpicHalt (
  input  wire        is_ebreak,
  input  wire [31:0] a0,
  input  wire [63:0] inst_cnt,
  input  wire [63:0] cycle_cnt,
  input  wire [63:0] data_cnt,
  input  wire [63:0] exec_cnt,
  input  wire [63:0] cyc_compute,
  input  wire [63:0] cyc_j,
  input  wire [63:0] cyc_ls,
  input  wire [63:0] cyc_csr,
  input  wire [63:0] inst_compute,
  input  wire [63:0] inst_j,
  input  wire [63:0] inst_ls,
  input  wire [63:0] inst_csr
);

`ifndef SYNTHESIS

import "DPI-C" function void ebreak();
import "DPI-C" function void good_trap();
import "DPI-C" function void bad_trap();

always @(*) begin
  if(is_ebreak) begin
    $display("INSTS:%d", inst_cnt);
    $display("CYCLES:%d", cycle_cnt);
    $display("IPC:%24f", $itor(inst_cnt) / $itor(cycle_cnt));
    $display("DATA:%d", data_cnt);
    $display("EXEC:%d", exec_cnt);
    $display("COMPUTE:%d / %d", cyc_compute, inst_compute);
    $display("JUMP:%d / %d", cyc_j, inst_j);
    $display("LD/ST:%d / %d", cyc_ls, inst_ls);
    $display("CSR:%d / %d", cyc_csr, inst_csr);
    if(a0 == 0) good_trap();
    else        bad_trap();
    ebreak();
  end
end

`endif

endmodule
```

sashimi-yzh, 2024-08-25:
可以学习一下宏的用法

sashimi-yzh, 2024-12-24:
这个问题与RTL编码相关, 与综合过程无关, 故关闭此issue.

closed 2024-12-24 by sashimi-yzh