OSCPU/yosys-sta #13: The support of yaml-cpp version

Stars-Collector, 2024-09-17:
The version0.8.0 of yaml-cpp was realeased last year,but it is not supported by yosys-sta still.And as I do as readme "apt install libyaml-cpp-dev",the 0.8.0 is what i downloaded.Do you have any plan to update the support version?

simintao, 2024-11-14:
We have use the yaml-cpp source code in the latest iEDA(https://gitee.com/oscc-project/iEDA/tree/master/src/third_party), so it is not need to install libyaml-cpp-dev.

junyu33, 2024-12-08:
One temporary workaround is:
- For arch users: `ln -s /usr/lib/libyaml-cpp.so.0.8.0 /usr/lib/libyaml-cpp.so.0.7.0` 
- For debian/ubuntu users: `ln -s /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.8.0 /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.7.0` 

However, this repo uses an iEDA binary from a downloading script, instead of using a git submodule to compile the source. Someone should write a script to update the binary to https://ysyx.oscc.cc/slides/resources/archive when a new version is commited in gitee source.
BTW, till 2024/12/8, I cloned a clean repo and tried `make sta`, it still relies on libyaml-cpp.dev.so.0.7.0
```
   Chip area for top module '\gcd': 920.360000

17. Executing Verilog backend.

17.1. Executing BMUXMAP pass.

17.2. Executing DEMUXMAP pass.
Dumping module `\GcdUnitCtrlRTL_0x4d0fc71ead8d3d9e'.
Dumping module `\GcdUnitDpathRTL_0x4d0fc71ead8d3d9e'.
Dumping module `\LtComparator_0x422b1f52edd46a85'.
Dumping module `\Mux_0x683fa1a418b072c9'.
Dumping module `\Mux_0xdd6473406d1a99a'.
Dumping module `\RegEn_0x68db79c4ec1d6e5b'.
Dumping module `\RegRst_0x9f365fdf6c8998a'.
Dumping module `\Subtractor_0x422b1f52edd46a85'.
Dumping module `\ZeroComparator_0x422b1f52edd46a85'.
Dumping module `\gcd'.

Warnings: 85 unique messages, 149 total
End of script. Logfile hash: 9b72cef776, CPU: user 0.54s system 0.06s, MEM: 33.45 MB peak
Yosys 0.23 (git sha1 7ce5011c24b)
Time spent: 59% 2x abc (0 sec), 7% 2x stat (0 sec), ...
./bin/iEDA -script /home/junyu33/yosys-sta/scripts/fix-fanout.tcl /home/junyu33/yosys-sta/example/gcd.sdc /home/junyu33/yosys-sta/result/gcd-500MHz/gcd.netlist.syn.v gcd /home/junyu33/yosys-sta/result/gcd-500MHz/gcd.netlist.fixed.v 2>&1 | tee /home/junyu33/yosys-sta/result/gcd-500MHz/fix-fanout.log
./bin/iEDA: error while loading shared libraries: libyaml-cpp.so.0.7: cannot open shared object file: No such file or directory
./bin/iEDA -script /home/junyu33/yosys-sta/scripts/sta.tcl /home/junyu33/yosys-sta/example/gcd.sdc /home/junyu33/yosys-sta/result/gcd-500MHz/gcd.netlist.fixed.v gcd 2>&1 | tee /home/junyu33/yosys-sta/result/gcd-500MHz/sta.log
./bin/iEDA: error while loading shared libraries: libyaml-cpp.so.0.7: cannot open shared object file: No such file or directory
```

simintao, 2024-12-08:
Yes, we need update the yosys-sta iEDA binary. However, you can build iEDA by source code from [https://gitee.com/oscc-project/iEDA/](https://gitee.com/oscc-project/iEDA), and replace the binary in yosys-sta.

simintao, 2024-12-08:
[image]

junyu33, 2024-12-08:
I prefer adding a symlink. It's a faster solution and it works well.