OSCPU/yosys-sta #14: 在生成verilog代码时候出现的错误

ccnuktd, 2024-11-13:
我在做ysyx总线的时候，想要测试多周期CPU的主频和性能，make sta后报以下错误。
```
Read Verilog file success : /home/ketted/Desktop/yosys-sta/result/top-MHz/top.netlist.syn.v
WARNING: Logging before InitGoogleLogging() is written to STDERR
I20241113 11:07:36.225592 28733 VerilogParserRustC.cc:41] load verilog file /home/ketted/Desktop/yosys-sta/result/top-MHz/top.netlist.syn.v
r str /home/ketted/Desktop/yosys-sta/result/top-MHz/top.netlist.syn.v
flatten module top  start
flatten module axi_lite_slave inst axi2mem_inst
flatten module sram inst sram_inst
flatten module \$paramod\handshake\COUNT=s32'00000000000000000000000010101000 inst ex2ls_inst
not found dcl stmt 2'b00
thread '<unnamed>' panicked at src/verilog_parser/verilog_data.rs:610:13:
not found connect net.
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 5
Aborted (core dumped)
./bin/iEDA -script /home/ketted/Desktop/yosys-sta/scripts/sta.tcl /home/ketted/Desktop/yosys-sta/example/top.sdc /home/ketted/Desktop/yosys-sta/result/top-MHz/top.netlist.fixed.v top 2>&1 | tee /home/ketted/Desktop/yosys-sta/result/top-MHz/sta.log
WARNING: Logging before InitGoogleLogging() is written to STDERR
F20241113 11:08:03.060506 28775 Sta.cc:79] File:/home/ketted/Desktop/yosys-sta/result/top-MHz/top.netlist.fixed.v is not exist.
*** Check failure stack trace: ***
Aborted (core dumped)
```
附上RTL设计和SDC文件
[npc.zip](https://github.com/user-attachments/files/17725632/npc.zip)
```
set clk_port_name clk
set CLK_FREQ_MHZ 400
if {[info exists env(CLK_FREQ_MHZ)]} {
  set CLK_FREQ_MHZ $::env(CLK_FREQ_MHZ)
} else {
  puts "Warning: Environment CLK_FREQ_MHZ is not defined. Use $CLK_FREQ_MHZ MHz by default."
}
set clk_io_pct 0.2

set clk_port [get_ports $clk_port_name]
create_clock -name core_clock -period [expr 1000.0 / $CLK_FREQ_MHZ] $clk_port
```
当前iED版本号：Git version: c02e5d3560c266c21dbc3c18309c5fc293324276

simintao, 2024-11-19:
commit c0481c4a5e3b85d14f8c8d790d17c06f4a5e436b 这个问题修复了，等王昊合入到OSCC

ccnuktd, 2024-12-05:
> commit c0481c4a5e3b85d14f8c8d790d17c06f4a5e436b 这个问题修复了，等王昊合入到OSCC

所以大概要什么时候才能合进去呢

sashimi-yzh, 2026-08-21:
目前已经完成了合并

closed 2026-08-21 by sashimi-yzh