OSCPU/yosys-sta #19: 多个 Yosys 指令不被支持

Clay-HHK, 2025-05-02:
## 问题概述

在尝试使用 `yosys-sta` 项目评估样例设计或个人 RTL 设计的时序与功耗时，我遇到了多个 **Yosys 指令不被支持** 的报错，严重影响流程使用。经过排查发现，这些问题均与 Yosys 版本不兼容有关，现汇总如下并提出改进建议。

---

##  遇到的报错（依次出现）

### 1. `autoname` 报错

ERROR: TCL interpreter returned an error: invalid command name "autoname"

### 2. `clockgate` 报错

ERROR: TCL interpreter returned an error: invalid command name "clockgate"

### 3. `check -mapped` 报错

ERROR: Command syntax error: Unknown option or option in arguments.

check -mapped

---

## 调试过程与尝试的解决方案

| 初始环境 | Ubuntu 22.04, 默认 Yosys 版本为 0.9
| 尝试版本1 | 克隆 GitHub 上 `https://github.com/OSCPU-Platform/yosys.git` 编译，依然缺失命令 |
| 最终解决 | 使用 `https://gitee.com/OSCPU/yosys.git` 编译的 Yosys 版本成功支持全部命令 |

---

## 问题分析

- 项目 `yosys.tcl` 使用了 `autoname`、`clockgate`、`check -mapped` 等非标准指令。
- 这些命令只存在于 OSCPU 修改的 Yosys 中，GitHub 仓库却没有包含实际可用源码（或没有文档说明）。
- 导致用户即使照着文档安装 Yosys，也仍然会因为版本问题报错。

---

sashimi-yzh, 2025-05-03:
这些是新版yosys的标准命令，并不是我们自行添加的，具体可参考[yosys的官方文档](https://yosyshq.readthedocs.io/en/latest/)。

初步推测是因为你没有按照README中对yosys版本的要求去尝试，导致使用了旧版的yosys。若按照README尝试后仍然遇到问题，欢迎继续讨论。

sashimi-yzh, 2026-08-21:
因长时间没有进一步讨论，关闭此issue

closed 2026-08-21 by sashimi-yzh